leandrolima-nyc/SnapmakerJ1 #1: Prusaslicer encountered an error

Learoy1, 2023-03-30:
[image]
[10736.gcode.zip](https://github.com/leandrolima-nyc/SnapmakerJ1/files/11116537/10736.gcode.zip)

encountered an error the first time I tried to slice as shown in the screenshot. I set up as your instructions so issue with your config

leandrolima-nyc, 2023-03-31:
[image]

leandrolima-nyc, 2023-04-02:
@Learoy1  Could you please confirm whether the issue has been fixed on your end?  If you encounter any further problems, please don't hesitate to reach out to me.  I'll be happy to assist you further.

Learoy1, 2023-04-02:
Thanks , installed 2.6 alpha 6 and that fixed this issue, but I have another issue with both post processing scripts not working

closed 2023-04-02 by Learoy1