leandrolima-nyc/SnapmakerJ1 #6: Bed not moved into position until preheat done

evilC, 2023-03-31:
It appears as if the bed is not moved into position until the preheat is complete.
With the cura profile, the preheat is initiated, then the bed is moved into position, then it waits for the preheat to complete.
This is much quicker

leandrolima-nyc, 2023-04-02:
I understand your concern.  However, with the relatively fast-moving speed that I have implemented, the bed moves into position quicker than the nozzle takes to reach the first layer printing temperature from the preheated temperature.  Therefore, there is no time wasted in the process.

On the other hand, if you cancel the job while the bed is in motion, it may cause issues with wifi printing and lead to the LCD freezing.  That being said, if you do need to cancel the job, there is a brief window of opportunity to do so.

Yet, upon further investigation, I have identified ways to improve the preheating process.  I value your feedback and will inform you of any progress made.

nickiaconis, 2024-08-06:
Over a year later, and I've just found this open issue while coming to make a similar suggestion.

I've been doing a manual preheat to about 150C for the nozzle (PETG) and 60C for the bed when using Luban, which generates Gcode that behaves similarly to the Cura profile as evilC described it. When doing so, the bed and nozzle will usually be fully preheated by the time the bed reaches the initial print height. I noticed that, with the current Prusa profile, the target temperatures of the nozzle and bed are set to 0C during the homing phase, which introduces a thermal loss to this manual preheating in addition to the time lost waiting for the bed and nozzle to come up to full temperature before raising the bed.

Additionally, the nozzle appears to be moved off of the silicone pad during preheating, which causes materials like PETG to ooze during the preheating phase, leading to splattering and under-extrusion during the priming phase, i.e. that initial line at the front of the bed, the skirt, etc. Admittedly, that's kind of what the priming phase is for, but I think it would be better to simply leave the nozzle parked on the silicone pad while preheating, if possible, to avoid the unnecessary oozing.

What were the ways you had identified back in April of last year to improve the preheating process? Would they address the concerns I've presented above? Is there anything any of us can do to help you make these improvements to the preheating process a reality?