leandrolima-nyc/SnapmakerJ1 #7: Bed not heating

evilC, 2023-04-02:
Generated G-code:

```
M82                                                 ;absolute extrusion mode
T0

  
    G1 X-5  F6000
    M104 S200 T0           ;preheat T0
    M190 S0 C3 W2    ;wait for bed temp
```

For some reason, it's setting `M190 S0`, even though I have set the first layer bed temp to 68 and subsequent layers to 60. I have no idea what `C3 W2` is - those options do not appear to be listed on the Marlin website

apphead, 2023-04-03:
Please export project as 3mf and attach it to your post.

evilC, 2023-04-03:
I switched back to Dylan's config for the print I was trying to do - can I just import your INI and have both imported?
I just did this, selected your printer config, and it happened again:
[100x100x10 block - Leandro.zip](https://github.com/leandrolima-nyc/SnapmakerJ1/files/11137571/100x100x10.block.-.Leandro.zip)

leandrolima-nyc, 2023-04-04:
@evilC, C&W are special arguments added by Snapmaker to quickly detect temperature.
C is the temperature diff
W is the detection window

leandrolima-nyc, 2023-04-04:
@evilC 
[image]

apphead, 2023-04-05:
@evilC your 3mf contains a misconfig, that explains the 0 bed temp
[image]

[image]

So, just select a correct filament profile and you will be good to go.

@leandrolima-nyc : 
So C is kind of the standby temp? Or diff to what?
W detection for what? 
Can you provide a reference to help me understand?

I might need to review my [enhancement](https://github.com/leandrolima-nyc/SnapmakerJ1/issues/9)

leandrolima-nyc, 2023-04-05:
@apphead I'm afraid I don't have any knowledge beyond the scope.  From what I understand, MacDylan originally obtained the information from the Snapmaker Cura profile and I received it from him. It appears to be something proprietary, so we're just following the established convention.  Your input and time are greatly appreciated, and I'm glad to see that improvements are being made. 

[image]

closed 2023-04-05 by leandrolima-nyc