leandrolima-nyc/SnapmakerJ1 #9: Enhancement: use idle_temp and standby_temp

apphead, 2023-04-03:
In 2.6-alpha 2 (I believe), new ways to handle ooooozing were introduced. One is a print setting called Ooze prevention, the other one is a filament property.   

My _Tool change G-code_  enhancement handles both scenarios. Could not find a combination of idle_temperature  and standby_temperature_delta settings for multi-material prints that returned a "bad" value.

```
{if next_extruder != current_extruder }
{if ooze_prevention==true}
	{if idle_temperature[current_extruder]<=1000} ;default value for idle_temperature is 2147483647, used if  idle_temp is deactivated
	M109 S{idle_temperature[current_extruder]} C3 W1 ;set filament idle_temp T{current_extruder}	
	{else}
	M109 S{temperature[current_extruder] + standby_temperature_delta} C3 W1 ;set print ooze_temp T{current_extruder}
	{endif}
{endif}

```

apphead, 2023-04-05:
Updated version (tested and passed)

{if next_extruder != current_extruder }
{if ooze_prevention==true}
	{if is_nil(idle_temperature[current_extruder])} ;use standby_temperature if idle_temp is not active
	M109 S{temperature[current_extruder] + standby_temperature_delta} C3 W1 ;set print ooze_temp T{current_extruder}
	{else}
	M109 S{idle_temperature[current_extruder]} C3 W1 ;set filament idle_temp T{current_extruder}	
	{endif}
{endif}

evilC, 2023-04-06:
Heh, I was just working on something similar.  
Does the same not apply to the initial preheat?

ie should the start gcode not be this?

```
M82                  ;absolute extrusion mode

T[initial_extruder]                   ;switch to initial extruder
M140 S[first_layer_bed_temperature]   ;start warming bed
{if total_toolchanges < 1}
M104 S[temperature]                   ;preheat nozzle
{else}
M104 S{if ooze_prevention && initial_extruder == 1}[idle_temperature_0]{else}[temperature_0]{endif} T0 ; Preheat extruder 0
M104 S{if ooze_prevention && initial_extruder == 0}[idle_temperature_1]{else}[temperature_1]{endif} T1 ; Preheat extruder 1
{endif}
```

evilC, 2023-04-06:
I think I have it

```
{if total_toolchanges < 1}
	M104 S[temperature]                   ;preheat nozzle
{else}
	{if ooze_prevention}
		{if is_nil(idle_temperature[0])}
			M109 S{temperature[0] + standby_temperature_delta} T0 C3 W1	; Set T0 temp to ooze prevention (Delta)
		{else}
			M109 S{idle_temperature[0]} T0 C3 W1 ; Set T0 temp to ooze prevention (Idle)
		{endif}
		{if is_nil(idle_temperature[1])}
			M109 S{temperature[1] + standby_temperature_delta} T1 C3 W1 ; Set T1 temp to ooze prevention (Delta)
		{else}
			M109 S{idle_temperature[1]} T1 C3 W1 ; Set T1 temp to ooze prevention (Idle)
		{endif}
	{else}
		M104 S[temperature_0] T0             ;preheat T0 to start temp
		M104 S[temperature_1] T1              ;preheat T1 to start temp
	{endif}
{endif}
```

evilC, 2023-04-06:
I think I may have it?  
```
{if total_toolchanges < 1}
	M104 S[temperature]                   ;preheat nozzle
{else}
	{if (initial_extruder == 0)}
			M109 S{first_layer_temperature[0]} T0 C3 W1 ; Set T0 temp to First Layer Temperature
		{if is_nil(idle_temperature[1])}
			M109 S{first_layer_temperature[1] + standby_temperature_delta} T1 C3 W1 ; Set T1 temp to ooze prevention (Delta)
		{else}
			M109 S{idle_temperature[1]} T1 C3 W1 ; Set T1 temp to ooze prevention (Idle)
		{endif}
	{else}
		M109 S{first_layer_temperature[1]} T1 C3 W1 ; Set T1 temp to First Layer Temperature
		{if is_nil(idle_temperature[0])}
			M109 S{first_layer_temperature[0] + standby_temperature_delta} T0 C3 W1 ; Set T0 temp to ooze prevention (Delta)
		{else}
			M109 S{idle_temperature[0]} T0 C3 W1 ; Set T0 temp to ooze prevention (Idle)
		{endif}
	{endif}
{endif}
```

evilC, 2023-04-06:
> Updated version (tested and passed)
>
> ```
> {if next_extruder != current_extruder }
> {if ooze_prevention==true}
> 	{if is_nil(idle_temperature[current_extruder])} ;use standby_temperature if idle_temp is [...]

Should this not use an M104? Yeah it's most likely not gonna wait because it's gonna be cooling down, not heating up, but better to be explicit?

closed 2023-04-25 by apphead